ut-robotics/picr21-team-bts #1: [Mechanics] - Test thrower sketch is not fully defined

Akustav, 2021-09-19:
The sketch is missing some constraints and dimensions, that is why it is blue.
[image]
If I were to move a vertex, unexpected things can happen, making it extremely difficult to modify this sketch further. Case and point: When moving one corner vertex this happened.
[image]

mbz4, 2021-09-19:
ok, will add some dimension pointers to the sketch to reference about origin

mbz4, 2021-09-23:
made a hot mess of things but it works now: sketch fully defined (i hope, tested a lot)

[image]

mbz4, 2021-09-23:
nevermind, just saw the blue peg hole xD
ok fixed!

Akustav, 2021-09-25:
Seems to be OK now with the components, yes.

closed 2021-09-25 by Akustav