ut-robotics/picr21-team-bts #2: [Mechanics] - Hole sizes should be correct to their purpose

Akustav, 2021-09-19:
Hole sizes in components are wrong. Holes that need to be threaded should have proper predrill diameters. They can be seen in this PDF

https://www.alliedmachine.com/AlliedMachine/media/AlliedMasterPageImages/Documents/Predrill-Diameter-Chart.pdf?ext=.pdf

If you want to have these holes threaded they are too large, if you want to pass a bolt through them, they are to small:
[image]

Similar issues here:
[image]

I suggest checking over all the holes in the components.

mbz4, 2021-09-19:
The hole sizes are actually correct.
1) the vertical holes are based exactly as the holes in the adapter plates of the previous teams design and as such i have no say on how they are setup
2) the horizontal holes that i added are not as constrained but the idea of printing the item is that you dont need to pretap in printing, you can tap the threads in later. I will reduce the 2.9mm diameter to 2.5mm 
3) the reduced diameter will allow for more 'meat' in the screws to be inserted after the printing is complete, in other words, i will manually add the threads by inserting the screws
4) if you prefer a more reliable attachment method, i could also use a nut, washer, bolt inlay technique or a jig saw resetter type interlock but this would increase the complexity of the preliminary design stage unnecessarily imho

Akustav, 2021-09-19:
1. For the vertical holes, the size is an issue. Tapped holes should be 2.5mm in diameter. Their location itself is OK if the frame has them there (missing from assembly).
2. I am not sure where you got the ideal hole sizes but usually, 3D-printed holes come out a little smaller than expected, requiring re-drilling for any precision at all. If the ramp holes are 2.9mm then the sidewall holes can not be 2.9 mm because a bolt needs to go through them, requiring a larger hole.
3. For component longevity it is suggested to cut proper threads. Threads "cut" using a screw are of lower quality since cutting is not clean and remains stay in the hole. It is OK if you fasten it only once ever but for a re-assemblable component, it should be properly cut. If you need more meat it can be adjusted in the slicer.
4. Bolts and threads are sufficient, it is just very unclear where holes need to be threaded and where not. This ambiguity is not good.

mbz4, 2021-09-23:
>> side adapter with motor mount hole sizes fixed: 3.1mm in passthru adapters, 2.5mm in chassis mounts
>> side adapter with no motor mount hole sizes fixed: same as above
>> ramp: tapped hole dim now corrected to 2.5mm

Akustav, 2021-09-25:
Seems fine now, yeah. Closing issue.

closed 2021-09-25 by Akustav