ut-robotics/picr21-team-bts #3: [Mechanics] - Thrower arch is weird

Akustav, 2021-09-19:
I am not sure why the throwers ramp is designed in such a way:
[image]
The radius is not aligned with the centre of the motor. The ball does not seem to make a lot of contact with the ramp at the beginning of the throw.

Additionally, the ball makes contactwith the motor alot sooner than with the ramp. This can force the robot to jump, reducing accuracy. This happens when the ball is pulled from position one to position 2:
[image]

mbz4, 2021-09-19:
Again, the measurements for motor position are only referenced from last years robot model.
In the given timeframe it was not possible to finetune the design to this degree.
Additionally, the ramp geometry is comprised out of two concentric circles, the motor axis is indeed not in the central axis mount as it should be. The draft angle between both ramp curves serves as the simplest self-centering geometry I could quickly implement.
One solution may be that I move the motor mount up a few mm such that the 'jump' would not occur, assuming the carpet is as depicted.

Akustav, 2021-09-19:
I think you should align the centre of the motor with the ramps curvature centre. Since you have full control of where the motor is located then you can also decide how far the ramp extends, helping avoid the issue where the robot is lifted during ball acquisition.

Using pro-paint skills I have illustrated some changed that would be nice:
[image]

Akustav, 2021-09-27:
Hey! Still a little bug here: 
[image]

Akustav, 2021-10-03:
Closing due to new design

closed 2021-10-03 by Akustav