ut-robotics/picr21-team-bts #4: [Mechanics] - 3D print orientation

Akustav, 2021-09-19:
In what orientation do you plan to 3D-print the central part? You need to take into consideration that one can not print on air. I am not sure if you considered it. It is not immediately clear to me.

mbz4, 2021-09-19:
1) https://www.explainthatstuff.com/how-3d-printers-work.html ...this is the thrower design review request, not print setup... for that i would first need slicer profile settings and slicing to take place...
2) the part is already designed in print orientation - the bottom most face allows for sufficient adhesive area such that the layers atop will not detach the piece off a properly calibrated buildplate
3) the part has no angles exceeding 60degrees in layering (see prusa references on design recommendations)

Akustav, 2021-09-19:
Ok. I guess this works. Just an additional thing that layer orientation in 3D printing matters, similar to how it is on wood. For this component the presented orientation most likely will work, just was not clear from the onset.

closed 2021-09-19 by Akustav