ut-robotics/picr21-team-bts #6: [Mechanics] - Intererence between components

Akustav, 2021-09-19:
There are components that sit inside one another. They can be seen below:
[image]

mbz4, 2021-09-19:
not sure i understand, but perhaps the joinery was not up to the task
i can offset the joints in the assembly if thats the issue

the adapter rings are there because i noticed and measured an offset of about 2mm either side
Im not sure what else would cause the interference but this is what you can expect in rushed designs.

Akustav, 2021-09-19:
The spacers seem to be placed wrong, they are 2 mm wide for a 1.7mm gap.

Akustav, 2021-10-03:
Clsoing to review new iteration

closed 2021-10-03 by Akustav