ut-robotics/picr21-team-bts #11: [Mechanics] - add chassis components to assembly

Akustav, 2021-10-03:
Currently the assembly has only the thrower. Add the important chassis components there to validate that there is no interference between components and that the attached throwers parameters match with clearance and other factors.

mbz4, 2021-10-03:
[image]

mbz4, 2021-10-03:
Chassis added to assembly. Attached thrower assembly's parameters match with clearances and other factors.

mbz4, 2021-10-03:
See to assembly file: "BTS TEST ROBOT ASSY.v3"

mbz4, 2021-10-03:
[image]