ut-robotics/picr21-team-bts #23: [Mechanics] - motor fastener impossible to lathe

Akustav, 2021-10-29:
All torque related to robot movement will be transferred through the fastening piece. The load is also dynamic since accelerations are quire strong on these pieces. I suggest making it out of metal (aluminum or similar) using the lathe.

mbz4, 2021-10-29:
[image]
Hey, the adapter piece is made of aluminum: I figured I could use the lathe to get the rough diameter then the mill to mill out the retention hole. Another solution is to simply use the lathe and make a monolith cylinder, with a side where the screw goes through to pin on the shaft of the motor, holes for the wheel screws and of course the hole for the motor shaft - not sure how to make the flat edge in that case but it should work even without it.
In the design, I actually use a flat face on the printed base and lid blocks to help translate the torques but these parts are not meant to work alone in this fashion. I wanted to distribute these loads as much as possible.

I guess I should remake the design to whatever we can fabricate then.
There should be something I can do to simplify this design.

mbz4, 2021-10-30:
[image]
Perhaps an adapter like this would be better then: 2 separate radii to lathe out, a centre bore for the motor shaft. Then on the mill: 5 vertical screw mount holes  (2.9mm dia), 1 cross-cut horizontal screw hole.

I went with a through cut on the horizontal screw mount in case we want to use part of a threadlocked screw insert as the flat face against which the motor shaft rests and/or as a second retention screw.
In any case it is an optional part that is simply a nice through-mill to fabricate.

mbz4, 2021-10-30:
I would sand the edges to get the chamfers except for the inner most one which can be achieved if I simply avoid lathing that part to get a nicer transition to the central shaft holder.

mbz4, 2021-10-30:
[image]

Akustav, 2021-10-31:
I would use the second variant, yes. One thing I think we can change is to reduce the OD of the holder. This lets us save material and reduce the time of making this piece. The 4/5/however many fastening holes can be pretty close to the edges of the shaft holder and the OD.

One suggestion I can make is that the holes holding the motor shaft should be offset from the wheel fastening holes, like so:

[image]

This reduces possible assembly complications in the case some the bolts used are too long

mbz4, 2021-10-31:
Seems the thread here is out of date: I addressed this issue soon after posting the above screenshots. It's angled correctly now, but to achieve this, I had to reposition the fastener holes in order to keep the horizontal retentors aligned with the flat of the motor shaft.

mbz4, 2021-11-02:
Why is this issue still open?

Akustav, 2021-11-02:
Current design is better.

closed 2021-11-02 by Akustav