ut-robotics/picr21-team-bts #25: [Mechanics] - Change way how rollers are retained

Akustav, 2021-10-29:
Currently the tolerances for the rollers might be to little and they might not spin freely. To avoid this you can increase side clearance on either side to around 0.2-0.3 mm and extrude a cone like area around the roller shaft that supports the roller from movement. The tolerances on the cone can be right at 4.0mm.

Visual illustrations (not to scale):

[image]

An example wheel picture

[image]

mbz4, 2021-10-29:
I can approach this issue by making the hole model for the rollers separate of the hole model for the shafts. I'll therefore define a new sketch on the side of one roller wall and then make a cut and then use the circular pattern on this cut for all the rollers.

mbz4, 2021-10-30:
Somehow missed the point about the extruded conical shape to keep the roller tight at 4.0mm while permitting play at 0.2-0.3mm in all new designs...
I think I can add it still but this will involve several hundred modifications.
Do you need me to fully assemble the wheels for each iteration or can I leave the non-final versions more disassembled (only 1 or 2 rollers fitted for illustration).
Can fully assemble once we've gone thru feedback I guess.

mbz4, 2021-10-30:
Okay, fixed for standard config. Tackling twin-bore config next.

[image]

mbz4, 2021-10-30:
[image]
It's just a conical extrusion, achieved thru a standard solid join extrude, then chamfered to a nice tip.

mbz4, 2021-10-30:
[image]
Done for both wheel configs.
Reviewed all issues again and I think it's all addressed adequately now.
Kindly accept the final design for submission per pending task requirements.

Akustav, 2021-10-31:
Seems nice now.

closed 2021-10-31 by Akustav