ut-robotics/picr21-team-bts #26: [Mechanics] - plans on covering the wheel

Akustav, 2021-10-29:
A question regarding how the wheel will fit into the whole assembly. Do you plan to cover the wheel?

This is important because currently rollers and bolts holding the wheel together can damage parts of the field (baskets) and opponents when they are grazed by the spinning wheel.

[image]

mbz4, 2021-10-29:
I thought we could inset the entire wheel-motor assembly inside the robot chassis such that this would not be an issue. With regard to this, I would prefer to just have the chassis encircle the entire assembly. If this is a new addition to the competition guidelines, I can make the screws sink into the wheel itself with counterbores and make the blocks thicker as well.

Akustav, 2021-10-31:
Having wheels in the chassis is reasonable. Just pointed out this to be careful with.

closed 2021-10-31 by Akustav