ut-robotics/picr21-team-bts #27: [Mechanics] New repatterned roller design based off feedback.

mbz4, 2021-10-30:
Didn't have a chance yet to repattern the twin bore config but this is a learning process so have patience.

[image]

Here's the newly repatterned wheel: diam is now 7cm, 20 rollers, 4 fasteners, lathable adapter, wheels intended for use inside the chassis. If this not acceptable, can extend base and lid blocks to encompass roller diam and have screw points in-bore.

mbz4, 2021-10-30:
[image]
I'll start work on a rough outline of the motor mount next.
Having an easily modifiable motor mount design series would help me with quick sanity checks in the wheel design department.

Also, I used the feature circ. pattern on a 8mm cut for all the roller shafts such that the resulting contour is tight fitting with no play. The roller bearing play is now 0.275mm either side of the roller. This is within the margins you specified. Even if 3dprinted, the shafts might be too constricted and the resulting assembly could deform due to printer tolerance mismatches.

mbz4, 2021-10-30:
[image]
Adjusted fastener holes to allow the horizontal fasteners enough space.
Aligned horizontal fastener with flat part of motor shaft.
Working on counterbores next.

mbz4, 2021-10-30:
Wanted to wrap up this assy before moving on to the twin-bore or the motor mount design daft.
[image]
Counterbores added: don't mind having them on either side of the wheel block because I thickened the wheel so that it now encompasses the entire roller diameter. This way there is 4mm of material from the bore to bore fastener hole to keep the parts clamped together. This shouldn't be an issue.
Will consider now this wheel assy done.

mbz4, 2021-10-30:
The counterbore mismatch was irritating so  I decided to have 2 distinct blocks instead such that the lid simply does not feature the counterbore for the fasteners.

mbz4, 2021-10-30:
[image]
Here's the twin-bore concept with 20 roller wheel assy, 10 rollers per base. Roller shafts fit in the contoured shapes nicely. The fasteners are a divisible of 20, but could have used 4 instead of 5 as well - 10 roller per wheel, 10 not divisible by 4 (thought process). The fasteners need to be about 30mm long, can cut to size if necessary. The rollers are fit inside the base blocks of the wheels. The entire motor mount assembly is at this moment assumed to fit the whole motor with wheel and mount inside the chassis perimeter.

Akustav, 2021-10-31:
I tried a quick prototype of a central unibody design:

[image]

Its diameter is 20mm and thickness of components is 4mm for the caps and 13mm for the center.

The center piece: 
[image]

Akustav, 2021-10-31:
But the layered approach itself is a little simpler to manufacture. I guess

mbz4, 2021-10-31:
I thought ~30mm is the minimum radius possible unless you use a second gearbox. 
Why this limit?
PC plate = 4.8mm
...need about 5mm off the bottom...
Need 'some' clearance for motor mount...
... motor shaft is offset additionally...
...putting the total at about 15mm radius minimum, resulting in 30mm diameter limiter.
I would add another margin there for fasteners and ease of installation.

Seeing how the top/bottom layouts intertwine, I think this could actually print just fine, just set the params for fine detail and ensure z res is sub 0.1mm
Bridging in your solution is not a problem either.
This would therefore work just fine but again, the diameter here might be too small unless an extra gearbox is added.
I would prefer to avoid the extra failure point and just make the wheel diameter at 30mm+ or how I set it at 'ball' diameter.
Do you want me to prepare a unibody config as well?

closed 2021-11-02 by mbz4

Akustav, 2021-11-03:
I am not sure what limitation you point at. I would make the wheel as small as needed to fit a set amount of rollers. Therefore the number of rollers decides the diameter, basically. As for printing, I am sure it is possible, just can be more fiddly since you are printing one roller slots upside-down with  the unibody.

As for height. The PC is 4,8mm but the motor can be recessed into the chassis. Similar to this:
[image]