ut-robotics/picr21-team-bts #30: Input required on every start of code

Akustav, 2021-11-02:
Current code asks for input on startup to configure basic properties like serial failsafe. This makes the startup of code more laborious since manual work is required. I would make it either a config file that can be adjusted based on requirements or have it as a startup cmd-line parameter if you need such things to be configured

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/motionModule.py#L47-L49

mbz4, 2021-11-02:
Can anyone explain why my troubleshooting routine found its way to """motionModule.py""" what sense does a troubleshooter have being there whatsoever...

Silaris, 2021-11-02:
As you see, the constructor exists, which uses default values. This thing is just for playing. There are 3 possibilities for config. One is to add a function to read it from the file. Another is just to write defaults inside code (base code can change request object values from this class). The third one is to use 0,0,0 as default and to calculate speed on spot. (probably will use 3)

Akustav, 2021-11-12:
Explained and expanded this issue in another one. Closing this.

closed 2021-11-12 by Akustav