ut-robotics/picr21-team-bts #31: Weird handling of motor speeds in the motion module

Akustav, 2021-11-02:
Currently the motionModule creates a request object on construction that is used to keep motor speed data. Would it not be more reasonable to send a Request to the MotionController that then parses this request data and sends the speeds, returning motor feedback as a Response?

I see you have a setSpeed function that sets the Request object parameters but why not give the Request object to the Controller directly?

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/motionModule.py#L21-L23

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/motionModule.py#L57-L75

Silaris, 2021-11-02:
You mean change self.req = Request()  to self.req = { a =1, b=2, c=3) in constructor?
Can do so. Creating additional classes, types, or interfaces for data models is my habit from TS and C#.
setSpeed function existence does not prove it will be used always. It is an option to set speeds as an array, not one particular value.

Akustav, 2021-11-02:
I would suggest using the Request instant for the motor_speed_val_input method, so it can be used something like this:

Request req = Request(speeds...)
motModule.mototor_speed_val_input(req)
#now the motor speeds are set. Technically it could also send the speeds right away, depends on what you want.

closed 2022-01-21 by Akustav