ut-robotics/picr21-team-bts #32: Uneven casing in python code

Akustav, 2021-11-02:
As was the case with C code so should python code be unified under one standard. Currently you have multiple syntaxes mixed.

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/motionModule.py#L29

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/motionModule.py#L47

mbz4, 2021-11-02:
What's the problem with that?

Akustav, 2021-11-02:
You use camelCase and snake_case at the same time. You should decide on one and use it.

Silaris, 2021-11-02:
Got you. It was 2 programmer code combination and we are using different styles D Will change

closed 2021-11-12 by Akustav