ut-robotics/picr21-team-bts #33: Motor speeds should be defined programatically

Akustav, 2021-11-02:
The current way how motor speeds are defined is weird is as in the user needs to provide input before it starts. This is pretty useless as code evolves. It would be more reasonable to define a set speed/direction inside code and drive based on that.

https://github.com/ut-robotics/picr21-team-bts/blob/f72089bd664bcdcd6df3d51b7ef4cdaa65b46ecc/Mainboard%20Communication%20Tasks/base.py#L14-L23

mbz4, 2021-11-02:
It is useless to whoever tries to use it apart from the guy ACTUALLY building the bot who just needs to access those few variables. Again, how did this get here in main commit?

Akustav, 2021-11-02:
I am not sure how this code got here but it is very narrow in scope and is unusable further in the project, therefore of low quality.

Silaris, 2021-11-02:
base3.py (current version) does not include this input. It is just for playing.

Akustav, 2021-11-02:
There are currently quite a few baseX.py files. Maybe it is reasonable to clean this up to have one base?

Silaris, 2021-11-02:
Will clean them up with a new camera controller. Now it is stored in base2.py with cv only and base3.py with realsense (faster to program in 1 file for test purposes)

Akustav, 2021-11-02:
Okay. I would use either folders to designate test code or keep them in a separate branch, else it can be difficult to understand what generically named files like base1, base2 and similar are supposed to do

closed 2022-01-21 by Akustav