ut-robotics/picr21-team-bts #35: [Mechanics] - change way roller shafts are retained

Akustav, 2021-11-02:
I would suggest changing the way how rollers sit inside the wheel housing. Currently you have 2 semicircles that enclose the shaft but you can encounter alignment issues due to the more complex matching between 2 moving semi-circles.

I would suggest a slot that is as deep as the shafts diameter and a flat surface retains this shaft in the housing.

A picture to illustrate:
[image]

mbz4, 2021-11-03:
Thank you for the suggestion, I will implement it because there are too many misalignment opportunities that could cause this. It may be possible to have a shape on the locking block to enclose them fully but that might again be asking for trouble.

mbz4, 2021-11-03:
I'll make a new design for the twin-bore that has a unibody mid section like your proposed solution.

Akustav, 2021-11-03:
Let me know when ready

mbz4, 2021-11-03:
Done!

Akustav, 2021-11-06:
OK

closed 2021-11-06 by Akustav