ut-robotics/picr21-team-bts #36: [Mechanics] - countersink style

Akustav, 2021-11-02:
Was wondering you you plan to use this style of countersink head or want to use, for example, a conical head?
[image]

mbz4, 2021-11-03:
We discussed this issue previously and in person: it would need 5 torx conical screws, ~30mm each. None are in stock at this length so for the twin-bore it is out of the question. We could apply them however to the regular wheel instead.

Akustav, 2021-11-03:
I think we can get the screws. Do I understand correctly that you want to use the twin-design?

mbz4, 2021-11-03:
We discussed that it would not be a problem if we keep both. But yeah, for now the ideal focus is to use the twin-bore (unibody) design.
The standard wheel design serves as backup in case the configuration misbehaves or something about movement dynamics...

mbz4, 2021-11-03:
If you are getting the torx screws, let me know their exact dimensions so I can model them in to test the fittings. I don't believe we have a problem in using either design, despite the 'ugly' counterbores.

Akustav, 2021-11-06:
https://www.tme.eu/en/details/b3x25_bn20039/bolts/bossard/3183910/ This would be a screw

Akustav, 2021-11-06:
I will close this task for now, I think the overall point was made.

closed 2021-11-06 by Akustav