ut-robotics/picr21-team-bts #37: [Mechanics] - drill through fastener if possible to simplify threading

Akustav, 2021-11-02:
I suggest threading through the wheel shaft mount to simplify threading:
[image]
Seems like one hole is harder to do since it collides with the crossing hole. Does it need a full-length hole?

mbz4, 2021-11-03:
From memory, I think they only need 2-3 mm, ~1mm short of the horizontal shaft fastener. We also discussed dropping down to 4 fasteners in this adapter case. That would resolve the collision issues as well as slightly reduce weight and complexity.
I want to use this adapter for the regular wheel configuration as well, despite the short displacement of fasteners from motor shaft.

mbz4, 2021-11-03:
Done!

Akustav, 2021-11-06:
OK

closed 2021-11-06 by Akustav