ut-robotics/picr21-team-bts #68: Threads don't have to be modelled

maidlam, 2022-01-12:
Might be better for general modelling performance to not have the threads modelled.
[image]

mbz4, 2022-01-12:
fixed on camera gantry but not on motor mounts yet

maidlam, 2022-01-13:
What I meant was that the threads in holes shouldn't be a 3D model, but when making a threaded hole, Fusion360 can show a texture instead.
There's an option in the "Hole" and "Thread" features to have the thread modeled. When there's no tick in the box, a texture will be shown instead.

The whole assembly should still show, which holes are threaded.

Picture is of the "Thread" feature in a motor mount side wall.
[image]

closed 2022-01-16 by maidlam