ut-robotics/picr21-team-bts #70: Check clearance of mainboard

maidlam, 2022-01-16:
I think there were some big capacitors on the mainboard.
Check the height of those capacitors to see that the mainboard fits behind the back motor.
Current distance between mainboard and motor is 12,2mm.
[image]

mbz4, 2022-01-16:
Steve assured me no Electrolytic caps are used in our mainboard design. Which means this could be a generous clearance margin. the 5mm standoffs are also optional. Just in case, added 3 extra mm of margin. Was trying to save space in case I fit the advanced thrower.

mbz4, 2022-01-25:
Given an entire finger fits after installation, I'd squarely say this issue is resolved. Closing.