ut-robotics/picr21-team-bts #71: Clearance issues with tenons

maidlam, 2022-01-16:
Check over all of the colored parts. Can cause fitting issues.
[image]

mbz4, 2022-01-16:
The motor mounts have a 0.05mm margin.
The Gantry features no margin - should I add the 0.05 to gantry mounts?

maidlam, 2022-01-16:
1. Motor mounts
The motor mounts holes are different dimensions - the tenons are 17,5mm, while their corresponding holes are 17,475mm wide. That is causing the interference there.
[image]
[image]

2. Gantry
The holes for the gantry don't need a margin, but the holes for the tenons are also out of alignment.
[image]

mbz4, 2022-01-16:
Motor mounts:
fixed on top chassis, removed clearance margin on bottom chassis (mounts remain fixed more often than on top chassis)
Gantry:
cannot identify where the misalignment occurred
Perhaps its a bad joint?

mbz4, 2022-01-16:
Joint is sound. Still cannot find cause for misalignment nor did I find any misalignment.

mbz4, 2022-01-16:
Fixed gantry assembly misalignment source.
To be sure, I extended the gantry to 118mm internal distance, applied to top and bottom chassis as well (battery cradle mounts had to be offset)

maidlam, 2022-01-16:
Still some issues with the bottom tenons of all the motor holders. The distance between the holes in the bottom plate is slightly bigger than the distance between the tenons.
Also, one of the holes in the top plate is missing.
[image]

closed 2022-01-17 by maidlam