ut-robotics/picr21-team-bts #74: Remove tabs to increase ease of assembly

Akustav, 2022-01-17:
For ease of assembly I suggest to remove the tabs on the back plate of the motor boxes. Otherwise a high level of precision is needed to attach the motors to the board and the top plate to the rest of the robot. Area is circled in red.

Additionally, you could technically skip the back- and sidewalls entirely. This reduces your mill, finishing and assembly processes. Your call.

[image]

mbz4, 2022-01-17:
thought it would help keep the mounts stable and reduce play during motion
also i like the design and spent quite some time on it
i agree on removing the top rear tab tho, it's really not needed
but i want to keep the side walls and back brace models just because :)

mbz4, 2022-01-17:
removed the top tab on the rear brace and the top chassis

[image]
lamenting the fact the bottom chassis looks kinda like a butt now xD

closed 2022-01-21 by Akustav