ut-robotics/picr21-team-bts #78: Utilize existing way of calculating driving speeds

Akustav, 2022-01-21:
I suggest using this method for most speed calculations:
https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/maneuver_inator.py#L27

It provides a steeper acceleration curve when objects are far away. This provides you with more responsive driving. But you need to keep a few points in mind:
* when the max speeds are too high the robot can overshoot the goal
* maxDeltaSpeed can be alot larger than maxAllowedSpeed, for example 200 and 40
* it takes quite a lot of trial and error to optimize for smooth driving

I am sure this will help you score more faster.

https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/game_logician_inator.py#L171

Silaris, 2022-01-23:
Do not remember, how this function works. Tried it, the robot was moving not as expected, moved the old function back. The quadratic scale inside it makes it difficult to fit our current linear formulas.

Akustav, 2022-01-23:
If your code is tuned for pure linearity, then nothing can be done at this point.

Below is a chart showing how speeds are calculated with different parameters:
sqrt = calculateRelativeSpeed(i, 100, 0, 300, 0, 50) with `normalizedDeltaFrac = sign * np.power(absoluteDeltaFrac, 2)`
lin = calculateRelativeSpeed(i, 100, 0, 300, 0, 50) with `normalizedDeltaFrac = sign * np.power(absoluteDeltaFrac, 1)`
normal = calculateRelativeSpeed(i, 100, 0, 50, 0, 50)

The difference is profound where max speed is held for more time and breaking occurs later.

[image]