ut-robotics/picr21-team-bts #79: Duplicate code in aim state

Akustav, 2022-01-21:
The way you determine if a ball is in frame is very redundant. You test both the X and Y values but how can a ball exist only in 1 dimension in a 2D image?

https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/game_logician_inator.py#L74-L88

Silaris, 2022-01-23:
Removed

ReikoR, 2022-01-23:
https://github.com/ut-robotics/picr21-team-bts/blob/2aaae7a75bbedf2f1caa16b1188a372ebb5a27ca/BTS%20dev/game_logician_inator.py#L111-L120

It's still not good.

Main point is that, if `self.ballY is None` is true, isn't `self.ballY is None` also true?

Something like this should be enough:
```python
if self.ballX is None:
    self.currentState = State.FIND
    return

curBallX = self.ballX
curBallY = self.ballY
```

Or you can use this if there are bugs in setting `ballX` to `None`:
```python
if self.ballX is None or self.ballY is None:
```

Silaris, 2022-01-23:
Fixed

closed 2022-01-23 by ReikoR