ut-robotics/picr21-team-bts #80: Weird exit for aim state

Akustav, 2022-01-21:
https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/game_logician_inator.py#L89-L93

This piece of code seems to exit the aim state when the basket is not in the frame? Why not instead try to find the basket inside the aim state since it is part of aiming?

Or am I misunderstanding?

Silaris, 2022-01-21:
Thanks. Forgot self.basketCenterX is -1 if no basket was found and this skipped step is skipped always. Will remove it.

Silaris, 2022-01-23:
Removed

ReikoR, 2022-01-23:
https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/game_logician_inator.py#L98

Variable divided by itself is 1. 

This line should probably be:
```python
side_speed = -base_speed
```

It's probably better to store last visible basketCenterX in a variable and use that to turn in that direction.

closed 2022-01-23 by Akustav