ut-robotics/picr21-team-bts #82: To different methods to solve the same problem in aim

Akustav, 2022-01-21:
https://github.com/ut-robotics/picr21-team-bts/blob/09e9488ed37ca8907c14233833fe658abc6a5cc7/BTS%20dev/game_logician_inator.py#L211

Why have the delay here? You are throwing for 40 frames times 0.05 seconds? The cameras framerate is at most 60, so every iteration has at least 16 ms of delay in it. Why not just increase the iteration number?

Silaris, 2022-01-21:
Parallel. We do not wait for the camera frame anymore. But this function will be adjusted letter. We haven't updated the throwing algorithm yet.

Silaris, 2022-01-23:
Updated throw method to additional aiming during the throw. Time. sleep remains. Reduced it to remove the delay.

ReikoR, 2022-01-23:
In general it seems sort of inefficient way to do it. Not sure if there is any benefit right now in running game logic more frequently than camera framerate. For minimum delay it would be better to run everything after new camera frames are received.

closed 2022-01-23 by ReikoR