ut-robotics/picr21-team-bts #87: Non-uniform code syntax

Akustav, 2022-01-25:
https://github.com/ut-robotics/picr21-team-bts/blob/3a1b7eb9764ac2e18287abd240885eefb3a9ea52/BTS%20dev/game_logician_inator.py#L139-L144

Currently camelCase and snake_case is used in a mixed way. It should be unified.

Akustav, 2022-01-25:
Here I can even see 3 different sytnax-es 

https://github.com/ut-robotics/picr21-team-bts/blob/3a1b7eb9764ac2e18287abd240885eefb3a9ea52/BTS%20dev/game_logician_inator.py#L324-L326

Silaris, 2022-01-25:
because of multiple class usage, all variables are now is camelCase (other classes had values in camelcase)

Akustav, 2022-01-25:
Some parts are still in snake_case:

https://github.com/ut-robotics/picr21-team-bts/blob/1e881e6a0b5c8d683954e1019223df77134a059e/BTS%20dev/game_logician_inator.py#L30-L33

https://github.com/ut-robotics/picr21-team-bts/blob/1e881e6a0b5c8d683954e1019223df77134a059e/BTS%20dev/game_logician_inator.py#L145-L148

It is also different file-by-file:

https://github.com/ut-robotics/picr21-team-bts/blob/1e881e6a0b5c8d683954e1019223df77134a059e/BTS%20dev/rc_inator.py#L22-L31

Silaris, 2022-01-25:
Fixed missing words in game_logician_inator.py. Am admitting, that other parts of code still contain different syntax but it might take a while to fix them.

Akustav, 2022-01-25:
Yeah true. I guess overall it is a small enough issue for the time being.

closed 2022-01-25 by Akustav