ut-robotics/picr21-team-bts #95: Weird angle adjustment system

Akustav, 2022-01-25:
. I would suggest having a separate bolt that slides in a similar arm that you have. It can be a little more offset from the ramp arms so it should not create too much interference.

The current solution has a lot of play and does not seem to be too elegant.

[image]

mbz4, 2022-01-25:
It is a functional design with lots of faults. So is the government. Closing with comment.

closed 2022-01-25 by mbz4

Akustav, 2022-01-25:
Re-opened issue due to there being no solution. Movement of the ramp is the core functionality and needs to work, else it might as well not exist.

mbz4, 2022-01-25:
fixed, added 3rd bolt

mbz4, 2022-01-25:
[image]
bolt length irrelevant, this works as you wanted
assign points - last call