ut-robotics/picr21-team-bts #97: How do you plan to hold the movable ramp arms in place?

Akustav, 2022-01-25:
Currently the arm is held in place by nothing. What would be used here?

[image]

mbz4, 2022-01-25:
Done.

Akustav, 2022-01-25:
I think the nut can completely be skipped if you make the 3mm hole a 2.5mm and have a sufficiently long screw. Nothing will bend the arms outwards. If you want to keep the 3mm hole, what nut will you use on the other side?

[image]

mbz4, 2022-01-25:
self tightening or a 4mm longer screw

mbz4, 2022-01-25:
+ washers

Akustav, 2022-01-25:
Can you use self-tightening nuts? The arm needs to move freely and nuts can not apply pressure to it.

mbz4, 2022-01-25:
the nut is tightened against the arm, not the adapter wall. A washer keeps the clearance.

Akustav, 2022-01-25:
Something like this? Red is where the spacer goes:
[image]

mbz4, 2022-01-25:
yea, was thinking 2 washers in case

Akustav, 2022-01-25:
But how do you clamp the nut down? Wont it push against the washer and arm, locking them in place?

mbz4, 2022-01-25:
then use a counter nut.