ut-robotics/picr21-team-bts #98: What holds the sensors and LED-s in place?

Akustav, 2022-01-25:
Currently nothing holds these components in place:

[image]
[image]

mbz4, 2022-01-25:
They are held there by sheer will.

mbz4, 2022-01-25:
Added retention feature bracket.

Akustav, 2022-01-25:
Currently the sensors can wobble in the retention clip due to excess space:

[image]

With LEDs it seems to be less of a problem because legs will be bent through the hole. This brings another issue, wont pins short themselves? Is there space for heatshrink or similar?

mbz4, 2022-01-25:
fixed

mbz4, 2022-01-25:
there is space for heatshrink there in addition to room for bending them each in their own direction

Akustav, 2022-01-25:
Heatshring itself has a wall thickness of around 0.3mm. If you do the minimum and shield 2 wires + some bending, it adds 0.6mm * 2 to the pins. Pins themselves are around 0.45 mm thick. The pin pitch on the sensor is 1.27mm, so in total the width of wires is at least 6.28mm wide and 1mm high. The design has space for 5 mm x 1.5mm

mbz4, 2022-01-25:
that's fine because the phototransistor does not need to stay flush on the face of the wall adapter

Akustav, 2022-01-25:
I think the space issue can simply be resolved by adding some chamfers:

[image]