ut-robotics/picr21-team-bts #99: 30mm bolts most likely too short and not rigid

Akustav, 2022-01-25:
IT seems to me that these bolts are too short and will bend inside soft plastic:

[image]

I suggest to use spacers.

mbz4, 2022-01-25:
True but I dont think a 40mm M3 bolt even exists ._.

Akustav, 2022-01-25:
Yeah, but there exists 40mm spacers that screw into the holes and don't apply all the force onto the hole. Spacers have a bottom that can rest on top of this component and the friction there transfers forces to the body of the component more evenly

Akustav, 2022-01-25:
Additionally, what fixes the nuts into the plastic? Threads don't bite into the bolt, that is why nuts are also torqued down. Currently the bolts just would not hold the middle ramp in a stable fashion.